ufo-kit/concert #355: Adding a callback to the @async future that prints an exception

MarcusZuber, 2015-01-30:
I had a few times problems that I havend noticed a exception of an asynchronous function, i.e. when 'device.motor.move' violates a limit.

I had the idea to add a function to every async as a callback that prints the exception if any accoured.
If this is also intresting for you, I could implement it in the near future.

tfarago, 2015-01-30:
Good idea, as long as the user can turn it off (sometimes you don't want to see the exception clutter) I would love to see a PR.

tfarago, 2015-02-03:
Would work for me...

MarcusZuber, 2015-05-21:
Is anybody aware of a possibility to get a new input-line in ipython?
After a print the cursor is behind the printed string. A '\n' is also only creating a new blank line.

MarcusZuber, 2022-02-01:
With the python3 asyncio coroutines this is not relevant anymore.

closed 2022-02-01 by MarcusZuber