ufo-kit/concert #363: Quantity that features measured value and target value

MarcusZuber, 2015-02-13:
I have a few cases in which I have a Quantity where I can set a target value and I can read a measured value, like in a Motor where I set the target position (then the controller tries to achieve the position) and I can read the real encoder position. 
In my most implementations the setter sets the target value and the getter reads the measured value. But this creates an (in my opinion) unwanted behavior, when using stash and restore.

My idea was to create a `ControlledQuantity` that features a target value (readable and writable) and a readable measured value. The stash/restore could then only restore the target values independently on any 'noisy' measured value.

Is there any elegant solution?

tfarago, 2015-02-13:
can you provide a use case for this? It will be easier to discuss on that.

MarcusZuber, 2015-02-26:
Ok. For example the x-ray tube.
`tube.set_voltage(120*q.kV)`
To reach the voltage setting it can take up to 1-2 minitues.
If I read `tube.voltage` in this time I will get someting inbetween the old voltage and the `120*q.kV` and if I use stash the unsetteled value will be stored. Also sometimes the controller is not able to set the selected value (like I'm always 1kV lower with the measured value). With using stash/resore multiple times I will deviate each time more from the target value.
Do you get my point?

tfarago, 2015-02-26:
Yes. Sounds reasonable and not too complicated to implement. I will come up with a PR at some point.

closed 2020-04-03 by tfarago