ufo-kit/concert #365: InjectProcess crashes on several occasions

tfarago, 2015-03-05:
When I call `wait` and then `start` again I get the following:
```python
from concert.ext.ufo import Backproject
from concert.coroutines.base import inject
from concert.coroutines.sinks import null
import numpy as np

images = [np.empty((128, 128), dtype=np.float32) for i in range(128)]
backproject = Backproject()
inject(images, backproject(null()))
backproject.wait()
inject(images, backproject(null()))
/usr/lib64/python2.7/site-packages/gi/types.py:43: Warning: invalid cast from `UfoCpuNode' to `UfoGpuNode'
  return info.invoke(*args, **kwargs)

(process:32532): Ufo-CRITICAL **: ufo_gpu_node_get_cmd_queue: assertion `UFO_IS_GPU_NODE (node)' failed

(process:32532): Ufo-CRITICAL **: ufo_gpu_node_get_cmd_queue: assertion `UFO_IS_GPU_NODE (node)' failed
Segmentation fault
```

When I `wait` right after `start`:
```python
from concert.ext.ufo import Backproject
backproject = Backproject()
backproject.start()
backproject.wait()
(process:734): Ufo-CRITICAL **: ufo_buffer_new: assertion `(requisition->n_dims <= UFO_BUFFER_MAX_NDIMS) && (requisition->n_dims > 0)' failed
Segmentation fault
```

Tested with current core & filters.

matze, 2015-03-05:
Failure of the second execution can be traced back to a mismatch of freed resources. The problem is that the resources (including OpenCL context etc.) are re-created for each pipeline execution but we keep the tasks lingering around which themselves may keep references to OpenCL objects. This is just asking for trouble.

Now, where to fix this? Ideally we would tear down everything and re-create everything from scratch. Alternatively, we could set up the arch graph containing the resources object and keep that around. Any preferences?

matze, 2015-03-05:
Your wait-after-start problem should be "solved" by [this commit](https://github.com/ufo-kit/ufo-core/commit/6d82029c51cf5fe688df7569baa860c4031c6b06). Although I think this is quite a pathological case. Usually you want to insert at least a single item.

tfarago, 2015-03-05:
> Failure of the second execution can be traced back to a mismatch of freed resources. The problem is that the resources (including OpenCL context etc.) are re-created for each pipeline execution but [...]

That smells like it has something to do with the memory leaks #338. Why are the resources recreated all the time? One context should be good for many runs AFAICS. I guess this is also the reason why `enable-reruns` still doesn't work (I am not sure, haven't tested in a while). 

> Ideally we would tear down everything and re-create everything from scratch

That probably won't work because how do you tear down a task which the user creates somewhere and wants to keep around?

> Alternatively, we could set up the arch graph containing the resources object and keep that around.

Would this mean that the OpenCL wouldn't be recreated all the time? If so I like this one because I think it will help with memory leaks. It's more state but seriously, what do I care when OpenCL itself has to keep the big fat `context`...

> Although I think this is quite a pathological case.

First, thanks. Second, it is and I think you shouldn't even to solve it that way. A *usable* error message would be fine instead of some strange dimensionality error which I have no idea about.

matze, 2015-03-05:
Your example is working for me and as far as I can see it is also not leaking more memory. If there is still an issue with that particular problem, please reopen.

commit d22c66fb (2015-03-05): Fix #365: use a single scheduler instance

Note that this requires ufo-core@274297 to work correctly.
closed 2015-03-05 by matze