ufo-kit/concert #368: Spectroscopic Detector

MarcusZuber, 2015-05-20:
I have a spectroscopic camera that creates multiple images in each acquisition. Each image contains the information of one energy channel. 
My idea to implement this camera was that the Camera.grab() function returns a stack of images, each slice containing one energy channel. To use this consistently I would implement a coroutine to select one energy channel and an adjusted ImageWriter addon. 

Would this be a suitable implementation? Are there any other dependencies on the grab() function?

matze, 2015-05-21:
I think this would be suitable. In fact the only guarantee about `grab()`s return value is that it is an ndarray. I think most important is that the dimensions are image dimensions (rows and colums mean x and y) not matrix dimensions (y and x).

tfarago, 2015-05-21:
What do you do with the images after acquisition? I guess you want to use them just like images from other cameras (3D reconstruction, ...). If that's the case then the processing stages after the acquisition which are not aware of 3D stacks (unlike e.g. numpy's `std` used by `focus` function)  must have a special implementation for this. This can be avoided if you think of every channel as a separate 2D camera. In this case you can use the rest of Concert without a single change. Or, maybe even better, one can have both, let's say I have a `SpectroscopicCamera` and I want to do CT on some channel, then one can implement `SpectroscopicCamera.get_camera_for_channel(50 * q.keV)` which returns (surprisingly :smile: ) a `Camera`, so you can use it normally by further processing.

MarcusZuber, 2015-05-21:
@tfarago I already thought about that, but then I get problems with the synchronization. I usually run them in trigger mode 'Auto'. I would need to use the SpectroscopicCamera for the trigger and the "SingelEnergy"-camera for the grab. 
My idea for the processing was to use the Specroscopic camera that returns these image stacks in experiment classes and then to attach consumers for each energy channel. To split the results I would use this slice-selcting coroutine.