ufo-kit/concert #380: Flat corrected backprojection is broken

tfarago, 2015-11-18:
* [ ] Segfautls
* [ ] Gives inconsistent results

matze, 2015-11-19:
I really have no clue. If I do it manually (i.e. first flat correct and then insert for the backproject) it works as expected, however, hooked up it together does not. Hmm ...

tfarago, 2015-11-19:
Can you upload the session or whatever you are using to gist so I can take a look and we have it linked to the issue please?

For now I'd say let's backproject without flat correction, with the new `Accumulator` I can easily flat correct myself before using `Backproject`. The reconstruction is not time critical in the next week's experiment so we can afford it.

matze, 2015-11-19:
> Can you upload the session or whatever you are using to gist so I can take a look and we have it linked to the issue please?

It's your (modified) check-fbpc session. I just ran this from the shell:

```py
result = Result()
fc = FlatCorrect(flat[SINO_POSITION], dark[SINO_POSITION])
inject((sino, ), fc(result()))
inject((result.result, ), bp(viewer())))
```

matze, 2015-11-20:
It does work on `ufo2` … so no clue why it's all messed up on `camlin`.

tfarago, 2015-11-20:
I just checked on ufo2 and current HEADs and when I execute `run` I get a nice result but when I call immediately after `run_no_fc` I get a segfault.

matze, 2015-11-20:
I just `zypper up`d UFO and it seems to work.

MariaMatveeva, 2016-09-28:
Was it fixed with the fix of #379?