ufo-kit/concert #382: Dimax start_recording stalls

tfarago, 2015-11-19:
When I use slow frame rate then after I call `dimax.start_recording()` it takes some time until the camera *actually* starts recording. The problem is, the uca camera state is only in stored in software so I don't have a real feedback from the camera (I hope I remember it right). Consequently, when I ask for a frame I get a timeout because the camera is not ready, e.g. when using [frames](https://github.com/ufo-kit/concert/blob/master/concert/experiments/imaging.py#L6).
I can imagine that libuca calls the appropriate function on the camera but doesn't ask for an ACK. @matze is it possible to get such an ACK? If not I don't see an elegant way to solve this, especially when the actual delay depends on the frame rate.

matze, 2015-11-19:
No, there is no such ACK or an API that tells me if the camera really *is* recording. I think the only sure way is to check that property `recorded-frames` is larger than 0.

tfarago, 2015-11-19:
What is libpco doing [here](https://github.com/ufo-kit/libpco/blob/master/src/libpco.c#L582) then? There is even a [public](https://github.com/ufo-kit/libpco/blob/master/src/libpco.c#L1279) API. Can I use it or is there something fishy?

matze, 2015-11-19:
Well, I call this function when doing `start_recording` so I suppose it reflects what I set. But if you want to be sure, just check/poll the `is-recording` property which maps to the public API.

tfarago, 2015-11-20:
I did, the camera simply returns status which is not real. Great. Workaround is in the PR.

PR #383 (2015-11-20): Add wait time to Dimax's start_recording
I know, but what else should we do?

fixes #382
commit 08244535 (2015-11-20): pco: Add wait time to Dimax's start_recording

fixes #382
closed 2015-11-20 by tfarago