ufo-kit/concert #384: Tab completion doesn't work

tfarago, 2015-11-26:
@MarcusZuber please provide additional description like Ipython version and when exactly it happens, we will take a look.

MarcusZuber, 2015-11-26:
I have ipython 4.0.0 since no older version is available in PyPi. Everything is installed globally (concert+ipython).
Here an example:
```
a=1
b=1
```

When I type an 'a' in ipython and go up I get 'a=1' as an input suggestion. In concert it just goes up in the history, so I will get 'b='1. I tested the same with a different account at the same machine and there both do the 'search-up' correctly. 
Afterwards I deleted the ~/.ipython directory in both accounts and created a new (empty) concert session and still I had the same behaviour. 

Is there somewhere else a (local) configuration for ipython or concert that could cause this?

matze, 2015-11-27:
Hmm, works for me. I just created a virtualenv with no access to system-site packages, installed IPython 4.0 and Concert HEAD. In that I could successfully tab-complete `a=1`. Could you
```python
from IPython import __version__
print __version__
```
and check that really IPython 4 is really used?

MarcusZuber, 2015-11-27:
It's 4.0.0 for both users

tfarago, 2015-11-30:
As suggested [here](https://github.com/ipython/ipython/issues/2516), installing `readline` via `easy_install` solves this.

closed 2015-11-30 by tfarago