ufo-kit/concert #406: PyplotViewer has issues with gobject imports

MarcusZuber, 2017-12-21:
The Pyplot(Image)Viewer is not working if an other gobject module was loaded before the viewer instance was created.

```python
# load something related to gobject
from concert.devices.cameras.uca import Camera
camera = Camera('mock')
from concert.ext.viewers import PyplotImageViewer
viewer = PyplotImageViewer()
```

This yields:
Process Process-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/usr/lib/python2.7/multiprocessing/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
  File "/home/ws/anka-bl-topo-to-0001/virtenvs/ufo/lib/python2.7/site-packages/concert-0.11.0.dev0-py2.7.egg/concert/ext/viewers.py", line 162, in _run
    from matplotlib import pyplot as plt
  File "/usr/lib/python2.7/dist-packages/matplotlib/pyplot.py", line 114, in <module>
    _backend_mod, new_figure_manager, draw_if_interactive, _show = pylab_setup()
  File "/usr/lib/python2.7/dist-packages/matplotlib/backends/__init__.py", line 32, in pylab_setup
    globals(),locals(),[backend_name],0)
  File "/usr/lib/python2.7/dist-packages/matplotlib/backends/backend_gtkagg.py", line 14, in <module>
    from matplotlib.backends.backend_gtk import gtk, FigureManagerGTK, FigureCanvasGTK,\
  File "/usr/lib/python2.7/dist-packages/matplotlib/backends/backend_gtk.py", line 16, in <module>
    import gtk; gdk = gtk.gdk
  File "/usr/lib/python2.7/dist-packages/gi/__init__.py", line 64, in __getattr__
    raise AttributeError(_static_binding_error)

AttributeError: When using gi.repository you must not import static modules like "gobject". Please change all occurrences of "import gobject" to "from gi.repository import GObject". See: https://bugzilla.gnome.org/show_bug.cgi?id=709183

Creating the camera after the viewer works fine.

Matplotlib: 1.5.1
pygobject: 3.20.0

MarcusZuber, 2018-01-16:
@matze: Could you have a look at this? I guess it's mostly related to glib and matplotlib.

matze, 2018-01-16:
Well, even though I know a bit of GLib and GObject, I doubt I can do anything about that. To me it looks like an issue with Matplotlib. You could try and change the backend from GTK to TkAgg.

tfarago, 2018-01-16:
I just updated matplotlib to the current version, didn't help. The Tkagg did though. Still not how it should be but at least it works.

matze, 2018-01-16:
Again, this is something I cannot fix because
```python
from gi.repository import GObject
import matplotlib.pyplot as plt
```
fails as well and has nothing to do with our software …

tfarago, 2018-01-16:
Yeah sure, I just wanted to inform about it.

MarcusZuber, 2023-01-20:
This issue is no longer present after moving to python3.

closed 2023-01-20 by MarcusZuber