ufo-kit/concert #424: Spyder api changend in version 3.0

MarcusZuber, 2019-03-15:
The spyder api changed when reaching 3.0 which made it incompatible to the current spyder session implementation.

The spyderlib package was renamed to spyder and the structure was changed.

`spyderlib.config` seams to be now `spyder.config.main`
`spydxerlib.app` seams to be now `spyder.app.mainwindow`

tfarago, 2019-03-20:
Please test it with the update-for-spyder-3-api branch. For me it works for both spyder 2 & 3.

MarcusZuber, 2019-03-22:
I get this error (spyder version 3.2.6):

Traceback (most recent call last):
  File "/usr/local/bin/concert", line 4, in <module>
    __import__('pkg_resources').run_script('concert==0.11.0.dev0', 'concert')
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 658, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 1438, in run_script
    exec(code, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/concert-0.11.0.dev0-py2.7.egg/EGG-INFO/scripts/concert", line 537, in <module>
    main()
  File "/usr/local/lib/python2.7/dist-packages/concert-0.11.0.dev0-py2.7.egg/EGG-INFO/scripts/concert", line 533, in main
    func(**vars(args))
  File "/usr/local/lib/python2.7/dist-packages/concert-0.11.0.dev0-py2.7.egg/concert/ext/cmd/spyder_plugin.py", line 44, in run
    spyder.run_spyder(app, Bunch(opts), [cs.path(session)])
  File "/usr/lib/python2.7/dist-packages/spyder/app/mainwindow.py", line 3021, in run_spyder
    main = MainWindow(options)
  File "/usr/lib/python2.7/dist-packages/spyder/app/mainwindow.py", line 283, in __init__
    self.open_project = options.open_project
AttributeError: 'Bunch' object has no attribute 'open_project'

tfarago, 2019-03-25:
Also when you delete build, concert.egg-info and dist directories *before* you re-install?