ufo-kit/concert #438: Seperate log files for single experiment executions

MarcusZuber, 2020-04-01:
I often put something like this in the `Experiment.prepare` to get a logger in the Experiment class, that creates one (additional) log-file for each experiment run in the folder of the current walker position, that has only the experiment related infos.

```
def prepare(self):
	self.logger = logging.getLogger("CTScan")
	handler = logging.FileHandler(self.walker.current + "/CTScan.log")
	self.logger.handlers = []

	formatter = logging.Formatter('%(asctime)s - %(name)s - %(levelname)s - %(message)s')
	handler.setFormatter(formatter)
	handler.setLevel(logging.INFO)
	self.logger.addHandler(handler)

	self.logger.info("Exposure time %s" % self._camera.exposure_time)
```

I could add this directly in the base Experiment class, if nobody sees a drawback by doing this.

tfarago, 2020-04-02:
I will do this in a little bit more general way, there will be a PR, we can discuss there.

tfarago, 2020-04-02:
I wouldn't touch the `prepare`, it should really stay virtual IMHO. We can put this stuff directly to `run` and let the experiment now in a constructor which devices to log, what do you think?

MarcusZuber, 2020-04-02:
Yes, the prepare was only my local implementation. It should be in the run, so the prepare stays virtual and can be implemented by the user.

When I use it, I do not really put the log of all devices in there, but call the logger in some functions like writing there all the experiment parameters and beam current etc. 

The logging of the devices is not very help-full at all times. (Most of it is like: LinearMotor::position = 1*q.mm without knowing wich of my linear motors it was). If a device would have a name this could be maybe improved a lot, but this would be a large change in the base device implementation.

PR #445 (2020-05-04): experiments: add per-run log handler
fixes #438
closed 2020-05-04 by tfarago