ufo-kit/concert #441: segmentation fault in libuca (or other external c-library) kills concert

MarcusZuber, 2020-04-09:
Sometimes a failing camera causes a segmentation fault in libuca (what I general should be in libuca, but the camera-drivers are often quite special). Since the uca-camera lives in the same thread as concert, concert is killed without having the chance of handling this error, or triggering some abort or cleanup functions.

I had two ideas how we could deal with this issue. To have something that survives and can still do something after such an event we need to have multiple processes.

First, we could move the whole uca in a separate process with an interface to it, that raises an exception when something in the uca process goes wrong. I think this will be quite ugly to implement.

The second idea was to launch a second process at the startup of concert, that check only if the "main" concert process is still alive and in case of a crash triggers an abort on all (pure python) concert devices.

tfarago, 2020-04-20:
I understand the need for this, but both solutions are quite dangerous (speed, busy waiting come to my mind). I think this should rather be handled on the libuca side. Or is that not possible?

MarcusZuber, 2020-04-20:
Yes, on the libuca side it would be much better. (Did you just volunteer?)
We would need to wrap everything with try-blocks in libuca (and ufo) to make sure, that issues in the drivers are only handled as exceptions and not as segmentation faults.

closed 2020-04-20 by MarcusZuber

tfarago, 2020-04-20:
Hahaaa, don't jump to conclusions! :wink: