ufo-kit/concert #448: Python3 Release Roadmap

tfarago, 2020-05-14:
Things to take care of:
- [ ] close [open pull requests](https://github.com/ufo-kit/concert/pulls)
- [ ] release last version for Python2
- [ ] code conversion to Python3.6+ (Python 2 **will not be supported anymore**)
- [ ] make tests work
- [ ] make CI work
- [ ] make sure docs are build
- [ ] make flake8 happy

What to use from and for Python3:
- [ ] async io instead of futures
- [ ] [imageio](https://pypi.org/project/imageio/) to unify image reading/writing

tfarago, 2020-05-14:
@MarcusZuber @sgasilov might be interested in the progress of this. Feel free to jump in with ideas about what to use from Python3 what is not in Python2.

MarcusZuber, 2020-05-14:
I think the asyncio will be the most intresting thing (and will also break the most old implemetations). I'm looking forward to see your first implementation of it 😃.

For the examples, I would propose to put them also in this repository. This will help to keep them more updated and they can be included in the tests as well.

Maybe we could also test other image-viewer implementations?

sgasilov, 2020-05-14:
Hello Tomas, hi Marcus,

hope you are both doing as well as you can. I really cannot contribute a lot because we are just starting to use Concert and, even if we have everything that we need for the first trials, no one can predict when we can try it in the real experiment. One comment perhaps to the that last idea of Marcus about image-viewer: Liza says that PyQTGraph is much faster than matplotlib pyplot. She also says though that documentation is meh and to work with it one would need to reverse-engineer available examples and use good pieces that you find. But maybe she can let to re-use her fragments of code, for instance, she has already done redraw, rescale, auto-adjust colorbar and such features for one project she is working on currently.

tfarago, 2020-05-14:
Yeah we have tried that but there were always some issues with it, that's why there is still no pyqtgraph-based viewer yet.

sgasilov, 2020-07-09:
Hello Tomas, 

I have confirmed in a series of tests that the entire Concert yield architecture does not work in a PyQT application. Lisa has time today to do some research and confirm that. She found this too:
"...
By default asyncio.get_event_loop() returns the default SelectorEventLoop. There are many functions internally in asyncio that make use of get_event_loop() to get the loop and attach callbacks to it. If you don't set_event_loop your custom event loop, all the Futures and coroutines get scheduled to the default event loop, which is never run.
https://stackoverflow.com/questions/32141623/pyqt5-and-asyncio-yield-from-never-finishes
Stack OverflowStack Overflow
PyQt5 and asyncio: yield from never finishes
I'm trying to create a new application based on PyQt5 and asyncio (with python 3.4, looking forward to eventually upgrade to 3.5 with async/await). My goal is to use asyncio so that the GUI stays
Try setting asyncio event loop to PyQt app event loop in __main__
loop = QEventLoop(app)
asyncio.set_event_loop(loop)
..."
in my naive interpretation the yield in the left side is in a separate even loop than yield on the right side so they never get connected in a Concert Consumer. It seems that with asyncio the fix is straightforward. 

What is your timeline? Should I wait until you convert Concert to Python 3? 
Direct "injects" structures  like this
inject((self.camera.grab() for i in range(10)), self.on_data_changed())
where consumer is a Concert coroutine which emits s pyqt signal
    @coroutine
    def on_data_changed(self):
        while True:
            im = yield
            self.data_changed_signal.emit("{:0.3f}".format(np.std(im)))
work in a PyqtThread correctly. However, does it make sense to continue development in 2.7 without using Experiment and trying to inject pseudo-acquisitions in Viewer and Writer without using Consumer?

Best,
Sergei.

tfarago, 2020-07-14:
Have you tried pyqt and concert with python 2? What you describe is all python 3 related.

sgasilov, 2020-07-14:
I have been working in Python 2.7.5,  PyQT5, and the latest Concert master. The on-line summary is that that the entire Concert architecture is probably going to work in PyQT5 with that one simple trick once future are converted to asyncio/await. I'm going to reach you on phone to chat about the details ))

commit 65498b12 (2021-12-17): Release concert 0.30

the first Python 3 release, fixes #448.
commit a733e8c5 (2021-12-17): Release concert 0.30

the first Python 3 release, fixes #448.
commit fe585eb1 (2021-12-17): Release concert 0.30

the first Python 3 release, fixes #448.
commit a1e7834f (2021-12-17): Release concert 0.30.0

the first Python 3 release, fixes #448.
closed 2021-12-17 by tfarago