ufo-kit/concert #449: Wrong experiment states after walker-exceptions

MarcusZuber, 2020-07-09:
If an exception occurs in in the Experiment.run() before  the try-block https://github.com/ufo-kit/concert/blob/e8eb389960fa9e7a26c11cf9a57256b3c7a9f0f3/concert/experiments/base.py#L249 the state in the state.variable is not correct anymore.

```
from concert.experiments.dummy import ImagingExperiment
from concert.storage import DirectoryWalker
walker = DirectoryWalker(root="/") # (any place where are no write permissions)
exp  =ImagingExperiment(num_darks=10, num_flats=10, num_radios=20, walker=walker)
f = exp.run()
```

The future `f`will stop and returns the exception `OSError: [Errno 13] Permission denied: '/scan_0001'` while the experiment state remains "running".

Moving all code in the try-block or checking the run()-future would fix this.

tfarago, 2020-07-14:
Is this supposed to change its state to error or just remain standby?

MarcusZuber, 2020-07-14:
I would say error is the right state.

commit cff8e607 (2020-07-20): experiment: ensure correct state on exception

fixes #449.