ufo-kit/concert #452: Devices are not deleted

MarcusZuber, 2020-07-31:
Something in concert keeps a reference to all devices (even until the session is closed).

Assuming such a session:
```
import logging
import concert
concert.require("0.11.0")

from concert.quantities import q
from concert.session.utils import ddoc, dstate, pdoc, code_of, abort

LOG = logging.getLogger(__name__)

from concert.devices.base import Device

class TestDevice(Device):
    def __init__(self):
        super(TestDevice, self).__init__()

    def __del__(self):
        print "__del__"

test_device = TestDevice()
```

When exiting concert noting is printed.

If I import this in a normal ipython session then "del" is printed when I exit the session.

This is especially a problem with uca-cameras, since the destructor in the plugin in not called (and in some cases the connection with the camera needs to be shut down in a proper way).

tfarago, 2020-07-31:
This goes very deep. How urgent is this?

MarcusZuber, 2020-07-31:
It is not very urgent (only annoying to physically switch detectors off and on when the session is restarted). 

But this could maybe be related to the "memory leak" we had a while ago.

MarcusZuber, 2022-02-01:
With the new python3 implementation this bug is no longer existing.

closed 2022-02-01 by MarcusZuber