ufo-kit/concert #454: Abstract class to run multiple (simmilar) experiments

MarcusZuber, 2021-03-18:
Having a base class for organizing loops over experiment executions (like scanning multiple samples or scanning one sample with different parameters) would be nice.
It would hold an experiment instance and can run it multiple times with the possibility to adjust parameters in between the executions.
@tfarago : In principle base class for the "director" we use in our sessions.

tfarago, 2021-03-31:
Yep, yep, yep, after python 3 switch.

MarcusZuber, 2021-12-20:
I put a simple implementation in https://github.com/ufo-kit/concert/pull/459.

My idea was, that the functions `_prepare_run(iteration: int)`,` _get_number_of_iterations()` and `_get_iteration_name(iteration: int)`
would be implemented in  an implementation of this base class.
Then the thing iterates up to the number_of_iterations and before the experiment is executed the` _prepare_run` does change settings or does something else to prepare the next execution of the experiment.

tfarago, 2021-12-20:
Then there should be also `_finish`, or? In the expriment we provide before and after flexibility, so it would make sense here too I guess.

MarcusZuber, 2022-05-19:
Implemented in https://github.com/ufo-kit/concert/pull/459 .

closed 2022-05-19 by MarcusZuber