ufo-kit/concert #456: Determine axis position, pitch and roll with align_rotation_axis without optimization

MarcusZuber, 2021-06-16:
Allowing no motor with `align_rotation_axis` would help to validate the alignment. Just skipping all the optimization loops when no motor is present would already do this.

tfarago, 2021-06-23:
Isn't that already possible? The `if [xz]_motor` looks promising to me:
https://github.com/ufo-kit/concert/blob/master/concert/processes/common.py#L390
https://github.com/ufo-kit/concert/blob/master/concert/processes/common.py#L401

sgasilov, 2022-01-31:
Tomas hello, 

Is z-axis the beam propagation axis and motor is samp roll/camera rotation in this case? (https://github.com/ufo-kit/concert/blob/master/concert/processes/common.py#L228-L230)  It seems like Y-axis and z-axis are is still inconsistent across the project.

tfarago, 2022-02-01:
x is the lamino or pitch axis, z is the beam direction or roll axis. Regarding consistency, you are right, I made a check box for us up in the description.

MarcusZuber, 2022-02-01:
Hi Sergey,
where did you find the inconsistency? I want to rewrite the concert.devices.positioners soon, where the coordinate system is also relevant.

sgasilov, 2022-02-11:
Hello Marcus, 

In tofu tomo z is the beam axis, but starting from tofu lamino and in tofu reco "y" is the beam axis and z is vertical axis. But here in Concert 3 in "auto-align" function the beam is again z axis. 
It would be good to stick to one notation: "y" axis = beam axis is most common.

Cheers,
SErgei.

MarcusZuber, 2022-02-12:
I also think y as beam direction is the most common.
I will try to make this consistent like this in the near future and I will also think if it is possible adding the functionality to define the "global" coordinate system in concert.