ufo-kit/concert #474: Software freezes in Experiment init if ctsetname is not a pattern and directory exists

sgasilov, 2022-03-17:
Hello, 

I understand that it is not how it is supposed to be used but users are users )). If ctsetname is hardcoded and directory exists the software crashes when Experiment init is called, most likely here:
https://github.com/ufo-kit/concert/blob/master/concert/experiments/base.py#L117-L121

Could you intercept exception or something in Concert or should I make a check at higher level? 

Cheers,
Serg.

MarcusZuber, 2022-03-18:
Hi Sergey,
what does ctsetname do? I guess it something that you use to define the walker-position.
Could you explain a little bit more how you set up and run the experiment class?

What I know (but didn't have the time to fix it) is, when you have a directory with some scan_XXXX folders in it. In the constructor the iteration is incremented until the first nonexisting scan_XXXX is found. If there are folders existing with higher values it will crash.

tfarago, 2022-03-18:
I make sure `name_fmt` is formattable now. About the other thing when you have high numbers in and start before, it starts on the lowest one and just increases. Then you will get a storage error from a writer if you go to a directory which exists already. We can also fix that and skip the existing ones if you want.

sgasilov, 2022-03-18:
Hello, 

basically we have Root dir, then ctsetname, which is supposed to be a pattern (e.g. scan_{<:03}, even scan{} will work in Python3) and then name_fmt which is applied to files. Typically we set the Root dir when change the sample and then, if there are multiple vertical views or time points, the Walker will create scan_001, scan_002, etc automatically (we never ever change name_fmt, although there is an entry for this). What some people can occasionally do is to hardcode the ctsetname - e.g. scan5. And then if this directory already exist init() will simply hang forever without returning anything.

tfarago, 2022-03-18:
I will also make the set_iteration smarter and merge.